damogranlabs/VS-Code-STM32-IDE #3: Build before download & run task

ali80, 2018-11-09:
If the project is changed, there is no point in downloading the old files to the MCU, so I think there should be an automatic build before download and run if any change is detected. 
this can be done by adding this to the download and run task in tasks.json file:
```json
"dependsOn":["Build project"]
```
the same is true for debugging so this should be added to the launch.json file:
```json
"preLaunchTask": "Build project"
```

schperplata, 2018-11-10:
I agree for debug launch configuration, added `build` dependancy task in the latest commit. 

_CPU: Download and run_ task will remain the same. The reason it is configured that way (eg. without previously executed build task) is that you can download some executable file just by manually copying it to the build folder and run task. That way IDE is more versatile, but demands more caution from the user. _Download and run_ task should do only those two things: download and run. 

Anyway, you are right, much more common use case is downloading the latest code. I added another _CPU: Build, download and run_ task.

I also updated comments and simplified task creation functions. Note that task creator will not overwrite user tasks (if json file is valid), so _update workspace task_ is safe to use.

closed 2018-11-10 by schperplata

ali80, 2018-11-10:
Maybe it would be appropriate to also make the build task the default task although its easy for the user to do so.