damogranlabs/VS-Code-STM32-IDE #5: #FeatureRequest Parallel building

Nostras, 2018-12-06:
Hi Domen,

I've tried your tool and it works pretty good so far.
Could I ask you to implement some sort of -j command in the makefiles somewhere?
I simply added the -j12 arg to the make command and that worked for me.
This command cuts my build times for egregiously cases from over a minute to less than 10 seconds.

schperplata, 2018-12-06:
Hello and thank you ;)
Adding new common switches to Makefile or 'make' command is generally not a good idea - someone might have a different configuration/CPU, different hardware or simply don't want this switch. In case of `-jX` command, X is number-of-cores-specific switch, meaning: I might have only 2 cores, while you have 4. I don't know what exactly will happen if this switch is not properly configured. So, no, this will not be added.
**Anyway, there are other simple solutions for your request:**
 - Manually modify generated tasks.json file in .vscode subfolder. 
Search for _Build project_ task and find: `"args": [ ]` field. Simply add your custom switches there, in your case this would be `"-j12"` added after `"GCC_PATH=...",` field. 
**NOTE:** This will be overwritten with next ideScripts update ('Update workspace' task).

 - Generate new build task.
Just copy entire _Build project_ task, rename it and add argument as described above.
**NOTE:** This task will NOT be overwritten on 'Update workspace' task. However, if something goes wrong (aka bad settings in any of other json files), your task will be overwritten (but stored in _.backup_ files).

- Modify task generator script.
Find _updateTasks.py_ and search for  `def getBuildTask(self)` function. At the bottom of this function, you can find:
`jsonTaskData["args"] = ["GCC_PATH=" + gccFolderPath]`
Just add your custom argument in the same manner:
`jsonTaskData["args"] = ["-j12"]`
**NOTE:** 'Make' command will be updated on 'Update workspace' task. This option will only be re-set to default on ideScript files update (eg.: future updates from this GitHub repository).

schperplata, 2018-12-09:
After a short break, I decided to give it a try and I have implemnted this feature.
Added '-jX' switch, where X is NUMBER OF CORES *1.5. **1.5** number is advised on many GNU make forum posts, so I go with this number.
On my 4-core PC there is not that much difference in time (about 20%), but I agree that every second you have must wait is a wasted second. 

Update your ideScripts.

closed 2018-12-09 by schperplata