damogranlabs/VS-Code-STM32-IDE #6: not working for me

jlhavens, 2018-12-20:
After instaling prereqs, I find the "First Steps" somewhat too sparse to follow.  No clear 
"Create CubeMX project and select 'Makefile' as output file" method.

I opted to open the example.  When i ran the update.py I got several  error similar to "Unable to find path to openOCD stlink.cfg configuration file on path:
        C:/Users/Admin/AppData/Roaming/GNU MCU Eclipse/OpenOCD/interface/stlink.cfg" where the scrips was looking for a file in a directory unlike the one I had entered to the earlier quiries.

After fixing these problems by cutting and pasting the appropriate files in what appear to be the non-standard directory locations, the script got past the "Tools paths updated." But went on to throw this  error
/////
Traceback (most recent call last):
  File "d:/Downloads/VS-Code-STM32-IDE-master/example/ideScripts/update.py", line 50, in <module>
    makefileData = makefile.getMakefileData(makeExePath, gccExePath)
  File "d:\Downloads\VS-Code-STM32-IDE-master\example\ideScripts\updateMakefile.py", line 86, in getMakefileData
    projectName = self.getMakefileVariable(makeExePath, gccExePath, self.mkfStr.projectName)[0]
  File "d:\Downloads\VS-Code-STM32-IDE-master\example\ideScripts\updateMakefile.py", line 296, in getMakefileVariable
    proc = Popen(arguments, stdout=PIPE)
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 997, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied
//////

Is there a solution to this problem?

Thanks, I'm sure a lot of work went into this, but its not working for me.
Jeff

jlhavens, 2018-12-21:
Thank you for your prompt response.

I loaded stm32CubeMX today.  Coming from the arduino environment, there is a lot that I am unfamiliar with in terms of the more professional programming tools.  After running through a youtube summery of the the CodeMX capabilities I am almost convinced that programming the STM32 is probably too complex for a hobby developer like me.

I'll give a try to your suggestions and let you know if I continue to have problems.

Thanks again,
Jeff

schperplata, 2018-12-21:
Oh, don't worry, CubeMX and STM32 products are one of the simplest and more user friendly products on the market. I've also started on arduino years ago, and although it took me some time, learning curve is still quite fast.
I advice you to start by the simplest example you can imagine, like blink an LED. Try to set up VS Code and than continue from there to more complex projects. Also, ST community is large and responsive, you will be doing amazing things in a matter of weeks!

Good luck!

closed 2018-12-21 by schperplata