damogranlabs/VS-Code-STM32-IDE #7: tasks cannot run with cmd.exe

bobby829, 2019-01-17:
My default terminal is using cmd.
I use cmd to run the tasks and get this.

```
'""C:' is not recognized as an internal or external command,
operable program or batch file.
The terminal process terminated with exit code: 1
```

I add an options in the in the tasks.json to make it run with powershell and it work perfectly.
```
        {
            "label": "Build project",
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "type": "shell",
            "command": "C:/Users/bobby/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.11-20180428-1604/bin/make.exe",
            "args": [
                "GCC_PATH=C:/Users/bobby/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/8.2.1-1.1-20190102-1122/bin",
                "-j6"
            ],
            "problemMatcher": {
                "pattern": {
                    "regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                    "file": 1,
                    "line": 2,
                    "column": 3,
                    "severity": 4,
                    "message": 5
                }
            },
            "presentation": {
                "focus": true
            },
            "options": {
                "shell": {
                    "executable": "powershell.exe"
                }
            }
        },
```

rreicher, 2019-01-17:
Hi,

Same error inside VSCode console

`> Executing task in folder NucleoF446RE_template: """C:/Users/romain reicher/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.11-20180428-1604/bin/make.exe" "GCC_PATH=C:/Users/romain reicher/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/8.2.1-1.1-20190102-1122/bin" -j6"" <

'""C:' is not recognized as an internal or external command,
operable program or batch file.
The terminal process terminated with exit code: 1`

`{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Build project",
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "type": "shell",
            "command": "C:/Users/name/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.11-20180428-1604/bin/make.exe",
            "args": [
                "GCC_PATH=C:/Users/name/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/8.2.1-1.1-20190102-1122/bin",
                "-j6"
            ],
            "problemMatcher": {
                "pattern": {
                    "regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                    "file": 1,
                    "line": 2,
                    "column": 3,
                    "severity": 4,
                    "message": 5
                }
            },
            "presentation": {
                "focus": true
            }
        },
        {`

schperplata, 2019-01-17:
Seems like your default integrated terminal is `cmd.exe`. Mine indeed is `powershell.exe`. I don't remember manually reconfiguring VS Code settings to use powershell, so I assume you either changed it manually or, _some other thing_ changed your default terminal.

As [Windows default terminal is now powershell](https://support.microsoft.com/en-us/help/4027690/windows-powershell-is-replacing-command-prompt), I decided to keep the tasks as they are, and ask user to switch to `powershell.exe`.
If you, for some reason, wish to keep the default terminal `cmd.exe` (instead of `powershell.exe`), you can reconfigure this VS Code workspace integrated shell path:
```
"terminal.integrated.shell.windows": "C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\powershell.exe"
```

It is possible to add this into workspace settings (as currently are `cortex-debug` settings), but I would prefer to keep as little modified VS Code settings as possible. I will implement it if there will be more requests.

closed 2019-01-17 by schperplata