damogranlabs/VS-Code-STM32-IDE #9: Build task : Error 1

VasenevEA, 2019-02-11:
Hello.
After run Build Task I see it in Terminal:
make: *** [Makefile:155: build/main.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:155: build/stm32f1xx_it.o] Error 1
The terminal process terminated with exit code: 1

What is that mean? Problem in my code?
How I can see any information about error path (line of code, like in CooCox IDE, or other)

schperplata, 2019-02-11:
It is hard to solve problems, if there is no sufficient information about the configuration/setup:
* What was the command string when running Build task?
* What did you try already? Did you try to compile a single file?
* Is _Update workspace_ task correctly executed?
* ...
It will be a lot easier if you describe your problem with as much detail as possible.

Anyway, a quick googling of this error code ([Make *** Error 1](https://stackoverflow.com/questions/5535548/make-error-1-error)) tells us, that this is actually GCC problem, not Make. Hence, check your GCC path and try to compile a single file by running Compile task on focused file. In my case, compile-time error is neatly displayed in terminal:
```
C:/Users/domen/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/7.3.1-1.1-20180724-0637/bin/arm-none-eabi-gcc -c -mcpu=cortex-m0 -mthumb   -DAVOID_EMPTY_DEFINE_FIELD_C -DUSE_HAL_DRIVER -DSTM32F051x8 -IInc -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Src/main.c -o build/main.o
Src/main.c: In function 'main':
Src/main.c:137:3: error: expected ';' before '}' token
   }
   ^
make: *** [Makefile:156: build/main.o] Error 1
The terminal process terminated with exit code: 1
```

VasenevEA, 2019-02-13:
This is crazy, but all problems is out when I was change GCC version:
from **8.2.1-1.3-20190202-1016** to **7.3.1-1.1-20180724-0637**

VasenevEA, 2019-02-13:
May be need to add this information to Readme?

schperplata, 2019-02-13:
Will investigate and add notes if necessary. 
Thank you.

closed 2019-02-13 by schperplata

poshcoe, 2019-02-13:
I've been getting the same issue - only when using the latest GNU-MCU-Eclipse GCC release (v8.2.1-1.3 20190202) on Windows. I believe this is to do with the following in the 'known issues' for this release:

> ## LTO & debugging on Windows
>The Arm 8-2018-q4-major release introduced a problem on Windows: enabling debugging information (-g/-g3) prevents -flto to properly link. For Release configurations it is not a problem to remove the debug options, but for Debug configurations this is unusable. **The workaround is to revert to the previous 7-2018-q2-update release, or to temporarily disable -flto.**

schperplata, 2019-02-19:
Thanks, added a note to a README.