damogranlabs/VS-Code-STM32-IDE #15: TypeError: expected str, bytes or os.PathLike object, not NoneType

dziurka, 2019-03-17:
When I execute update.py script, the following errors appear:
`Traceback (most recent call last):
  File "ideScripts/update.py", line 43, in <module>
    buildData = bData.prepareBuildData()
  File "/home/dziurka/STMProjects/IDE-TEST/ide/ideScripts/updateBuildData.py", line 87, in prepareBuildData
    buildData = paths.verifyExistingPaths(buildData)
  File "/home/dziurka/STMProjects/IDE-TEST/ide/ideScripts/updatePaths.py", line 102, in verifyExistingPaths
    buildData[self.bStr.gccInludePath] = utils.getGccIncludePath(buildData[self.bStr.gccExePath])
  File "/home/dziurka/STMProjects/IDE-TEST/ide/ideScripts/utilities.py", line 395, in getGccIncludePath
    folderPath = os.path.dirname(filePath)
  File "/usr/lib/python3.6/posixpath.py", line 156, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
`
Any ideas?

schperplata, 2019-03-17:
Hmmm. I would need a little more info. Can you share your ```buildData.json``` from this workspace .vscode folder and ```toolsPaths.json``` from ```"$HOME/.config/Code/User/")```?
Or ideally, just pack whole project?

Also, are you using the latest ideScripts version?

dziurka, 2019-03-17:
Thanks for the quick response. I downloaded `ideScripts` directly from the git.
`builtData.json` does not have any data and `toolsPaths.json` does not exist.
Link to the project: [link](https://drive.google.com/file/d/1xWAks2yi8YmUBf50dG-cuPVlCyA91G8y/view?usp=sharing).

My all paths:
`1. /home/dziura/STMProjects/VSCODE/arm-none-eabi-gcc
2. /usr/bin/make'
3. /usr/bin/openocd
4. /usr/share/openocd/scripts/target/stm32f4x_stlink.cfg
5. /usr/share/openocd/scripts/target/stm32f4x.cfg
6. /home/dziura/STMProjects/VSCODE/STM32F40x.svd`

`STM32F40x.svd` and `arm-none-eabi-gcc` are downloaded from the pages indicated in readme.

Command to run script:
`dziura@dziura-nb ~/STMProjects/IDETEST/ide $ python3 ideScripts/updatePaths.py `

My OS is Solus Linux.

schperplata, 2019-03-17:
Since I don't use Linux, maybe @poshcoe could tell more about `gcc` and `make` paths. 

Anyway, I assume here is the problem:
```python
 gccExeFolderPath = os.path.dirname(gccExePath)
 gccFolderPath = os.path.dirname(gccExeFolderPath)
 searchPath = os.path.join(gccFolderPath, "lib", "gcc", "arm-none-eabi")

 fileName = "stdint.h"
 filePath = findFileInFolderTree(searchPath, fileName) # returns None
 folderPath = os.path.dirname(filePath) # fails
```
As it seems, `findFileInFolderTree()` is unable to find `stdint.h` file in your GCC directory. Is this file really accessible? Did you manually change/rename GCC folder structure?

dziurka, 2019-03-17:
You are absolutely right. I did not include all directories for gcc. My fault. Thank you for your time. 
Topic to close.

closed 2019-03-17 by dziurka