damogranlabs/VS-Code-STM32-IDE #17: #FeatureRequest Support for several contributors

dziurka, 2019-03-18:
Are you planning to add support for several contributors?
At the moment, paths to additional sources (.c / .h) are kept in one json along with individual paths to user devtools.
This makes it impossible to work together on the code. The next commits would overwrite the paths to devtools.
In theory, a separate json file with paths to the sources would be sufficient.
Despite it all - huge thanks for a great tool!

schperplata, 2019-03-19:
I indeed wasn't thinking about multiple contributors when I started this project. That is also the reason why `buildData.json` file has `.backup` extension, since several paths make problems and I don't like it.
It is not the `.c/.h` files that are problematic, since this paths are fetched from `Makefile` and `c_cpp_properties.json`, and they are usually relative to `.code-workspace` folder. Paths to configuration files and tools executables (`.svd`, `.cfg`, `make`, `gcc`) are the one that causes problems when sharing code.

Until I came with a better idea how to handle/cache this paths, a workaround is to simply delete this keys from `buildData.json` and run Update task. If a valid `toolsPaths.json` exist on your PC, paths to tools (`gcc`, `make`, `openOCD`) will be fetched from there. Anyway, configuration files (`.svd` and `.cfg`) will still be wrong and would need to be updated.
Keys to delete (basically everything below "buildDir" key):
* "gccExePath"
* "gccInludePath"
* "buildToolsPath"
* "pythonExec"
* "openOcdPath"
* "openOcdInterfacePath"
* "targetExecutablePath"
* "cubeMxProjectPath"
* "openOcdConfig" (*)
* "stm32SvdPath" (*)

Thank you. :)

schperplata, 2019-04-01:
Update!
Now tools paths in 'buildData.json' are always fetched from user (system) 'toolsPaths.json', and are overwritten on each Update task. Build data still stores both, tools paths and workspace target configuration paths. Target configuration files are now local to workspace - copied to '.vscode' folder. 

Once you get a workspace from a different contributor, just run Update task and tools paths should be updated to your local paths.

closed 2019-04-01 by schperplata