damogranlabs/VS-Code-STM32-IDE #18: osx support

om2kw, 2019-04-01:
many thanks for nice projects, works fine on OSX besides few changes:
in utilities.py add
if platform.system() == "Darwin":
            osIs = "osx"
return osIs

.....
osIs = detectOs()
    if osIs == "windows":
        vsCodeSettingsFolderPath = tmpStr.defaultVsCodeSettingsFolder_WIN
    elif osIs == "unix":
        vsCodeSettingsFolderPath = tmpStr.defaultVsCodeSettingsFolder_UNIX
    elif osIs == "osx":
        vsCodeSettingsFolderPath = tmpStr.defaultVsCodeSettingsFolder_OSX
....
if osIs == "unix" or osIs == "wsl" or osIs=="osx":  # detected unix based system
        pythonExec = "python3"

in templateStrings.py:
defaultVsCodeSettingsFolder_OSX = os.path.expandvars("$HOME/Library/Application Support/Code/User/")

om2kw, 2019-04-01:
and in updateTasks.py
 elif osIs == "osx":
            openCubeCommand = "/Applications/STMicroelectronics/STM32CubeMX.app/Contents/MacOs/STM32CubeMX"

schperplata, 2019-04-01:
Thank you, nice to hear you find it useful.
Anyway, I really don't like adding stuff that I can't test on my own - I only work with Windows, hence, some more input is expected from you :) For example, docs for `platform.sytem()` does not mention `Darwin` and I think it would be nested here, under 'posix' tag:
```python
...
elif os.name == "posix":
    release = platform.release()  # get system release
    release = release.lower()
    if release.endswith("microsoft"):  # Detect windows subsystem for linux (wsl)
        osIs = "wsl"
    else:
        osIs = "unix"
    # additional magic here?
```
... and so on. 

Now, I suggest, you modify files as you think is the best solution for your problem (since you obviously already did it) and make a pull request. I will than check and if everything looks OK, I will merge it to this repo.

closed 2019-06-01 by schperplata