damogranlabs/VS-Code-STM32-IDE #20: New location of .SVD and .cfg files

noppej, 2019-04-25:
Hi, I notice you now copy .SVD and board .cfg files to the .vscode folder. Is there a specific reason for that? I used the previous versions of your scripts (before you copy these files) very successfully, and found it cleaner to reference these files in their proper location. 

If the .cfg file is updated, which one do we need to update, and will the update.py file re-copy the updated file?

schperplata, 2019-04-25:
Yes, this was done intentionally to [enable multiple code contributors](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/17). This concept allows us to share workspace without redefining target-specific paths on each _fetch/update_ from other contributor. 

If svd or cfg files are changed (which is unlikely, but possible), I would:
* update OpenOCD/Keil files in default installation (%userprofile%\AppData\Roaming\GNU MCU Eclipse for windows)
* delete this files in project .vscode folder and run Update task.

noppej, 2019-04-25:
Perfect. I can live with that explanation. Would you like some help updating the docs?

PS. Thanks again for your work here. Being able to develop in VSCode is such a pleasure after being stuck with Eclipse based IDE's for STM for so long.

schperplata, 2019-04-25:
> Would you like some help updating the docs?

Of course, If you find anything unclear, or have a better idea, go ahead and contribute! Once you create a pull request, I will just quickly check if everything is OK, and than merge. 
You are very welcome to do so :)

Thanks :) I am also a fan of VS Code and I use it on all projects now!

noppej, 2019-04-27:
Thanks ... will do :)

closed 2019-04-27 by noppej