damogranlabs/VS-Code-STM32-IDE #21: texane/stlink support

Goldenstarc, 2019-05-19:
I'm using texane/stlink in osx.
Correctly osx is not on os type.
add this after that.
thanks.

schperplata, 2019-05-23:
Please [see this issue and my comment regarding osx support.](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/18)

Anyway, please, when you open an issue, write it in a way me and others can understand it. Currently I don't even know what the real problem is and I don't plan spending time debugging unclear issue reports. :)

schperplata, 2019-06-01:
Update: [osx support added by _om2kw_](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/18), please check if your problem is still relevant.

Goldenstarc, 2019-06-02:
Sorry for that, you’re right.
This program uses OpenOCD while I requested to add texane/stlink support.
 Another this is that after setting include path, i also have to define uC model in 
'c_cpp_properties.json’ to fix vscode issues

schperplata, 2019-06-03:
> This program uses OpenOCD while I requested to add texane/stlink support.
As I only use stlink, currently there is no plan to extend support for other tools. It can be done for sure, but quite a lot of work is needed to port all configurations, tasks and other features to support other tools. 

> Another this is that after setting include path, i also have to define uC model in
'c_cpp_properties.json’ to fix vscode issues
I am not sure I understand what you mean or what is the problem. Of course you must set inlude paths, so VS Code, GCC and other tools know where to search for files. And if your project (code in files) demands some predefines, than of course, you also need to add this to 'c_cpp_properties.json'

Goldenstarc, 2019-06-03:
> I am not sure I understand what you mean or what is the problem. Of course you must set inlude paths, so VS Code, GCC and other tools know where to search for files. And if your project (code in files) demands some predefines, than of course, you also need to add this to 'c_cpp_properties.json'

[image]

schperplata, 2019-06-03:
Now I see. Your issue has nothing to do with this project. 

Please see the main page, docs and examples.

closed 2019-06-03 by schperplata