damogranlabs/VS-Code-STM32-IDE #22: Problem working with importKeilProject.py

Edu-Ape, 2019-05-23:
Hi, first of all, thank you very much for so nice project.
I trying to import a Keil Project into VS Code, using importKeilProject.py.
When it is calling to CubeMX, shows this error:
"CubeMx returned non-zero exit code. Something went wrong: Error: opening registry key 'Software\\JavaSoft\\Java Runtime Environment'\r\nError: could not find java.dll........
So, I use same parameters that importKeilProject.py is using to call CubeMX, directly from command line, and the Makefile template is created without any problems.
So my question is, why inside VS Code, importKeilProject did not found java?
Am I doing something wrong?
Sorry I'm newbie trying to learn. Thank you again for your time and support
VS Code version: 1.34.0
Java version: 1.8.0_201
Python version: 3.7.3
CubeMX version: 5.1.0
Windows 7 Ultimate service pack 1

schperplata, 2019-05-23:
Hard to say what is wrong. Did you inspect error message carefully?
`"CubeMx returned non-zero exit code. Something went wrong: Error:` is my message, while the rest is actually CubeMX returned status:
`opening registry key 'Software\JavaSoft\Java Runtime Environment'\r\nError: could not find java.dll........`
Do you have some custom java installation?
I would suspect that windows 7 should work, but I never tried, I've always worked with win 10.

Edu-Ape, 2019-05-23:
Thank you for your fast answer!
I debugged your phyton code, and copy the command issued on file 288 of importKeilProject.py.
I opened a new cmd Terminal on VS Code, and run the command. All worked Ok and finished with Makefile done.
I forgot mention you I'm using windows 7 64 bit, and the phyton version I sent you was 32 bit, so, I will check to use all tools versions in 64 bit first. I'll keep you posted

Edu-Ape, 2019-05-24:
Finally, seems it was a conflict between VS Code and Python versions. I uninstalled Python completely and installed only Python version 3.7.3 64 bit and it worked.
Now, I'm having another error but could be from the Keil project I'm trying to import.
The error is "Multiple '*.s' source files listed...", I have to check if the project has more then one startup file.
I will not close the comment for a while in case you have some suggestion to give me.
Thank you again for your support

Edu-Ape, 2019-05-24:
I forgot to mention, besides, I had to specifically configure the python path on settings.json (Python: Python Path)

schperplata, 2019-05-24:
If VS Code doesn't automaticaly recognize installed Python(s), the reason could be that you are having python installed on some custom path, or python.exe can't be found on system PATH.

Multiple startup

Edu-Ape, 2019-05-27:
Hi, I was tracking the error "Multiple '*.s' source files listed..." and I found the following:
I'm trying to import the Keil project from https://github.com/RAKWireless/RAK5205-WisTrio-LoRa
When 'importKeilProject.py' scan the file on row 187, as soon as it found the file ReleaseNotes.txt on 'doc/Firmware' directory, it start to append all files again and finally it finish with the double of c files and two startup.....s files.
So, after that, it stops with the error message mentioned before. Am I doing something wrong? Is the Keil project file bad formatted?
Please, let me know your opinions. Thank you

Edu-Ape, 2019-05-28:
I think the recursive nature of getElementsByTagName is producing nodes duplications from the Keil .uvprojx file, on row 184 of that project.

schperplata, 2019-05-28:
Given Keil project is a two target workspace - all files are present in both target solutions, hence _importKeilProject.py_ script has all duplicated. 
[image]
By removing one of the targets from Keil workspace (and save), this script will work properly. 

Currently such more complex Keil projects are not handled - and probably won't be, since there is whole bunch of other problems for multi-target Keil projects.

Edu-Ape, 2019-05-28:
Thank you very much for your time!
Ok, I understood. Are those pictures coming from Keil IDE? Because I don't have Keil licence to manage this project, that's why I'm trying to use for all my projects the VS Code.
Again, thanks for your understanding

Edu-Ape, 2019-05-29:
Finally, with your considerations and kind support, it worked. Now I will continue with the rest of the project. Thank you

closed 2019-05-29 by Edu-Ape