damogranlabs/VS-Code-STM32-IDE #24: Invalid path detected in 'buildData.json' 'stm32SvdPath' key

niekez, 2019-06-18:
Getting error:

`Invalid path detected in 'buildData.json' 'stm32SvdPath' key`

I need to re-enter Svd path every time I run update.py. The script copies the svd file to .vscode and adds the path local path to buildData.json. Changing the path in buildData.json to the initial absolute path doesn't work.

On Archlinux

schperplata, 2019-06-19:
Can you share your environment?

Changing buildData.json is not the way to go - you can update all the relevant paths by running updatePaths.py script. What happen if you:
* delete .svd file in .vscode folder
* delete/clear "stm32SvdPath" key in buildData.json
* run update.py  

Is the path you give 100% correct? (It might be, that .svd file is left there from previous run, and you don't have a valid path (like some hidden characters/spaces/?)

niekez, 2019-06-20:
My environment:

```
Version: 1.35.1
Commit: c7d83e57cd18f18026a8162d042843bda1bcf21f
Date: 2019-06-14T21:57:30.073Z
Electron: 4.1.4
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.30-electron.0
OS: Linux x64 5.1.9-arch1-1-ARCH

```
I've followed the steps above. It copies the svd, filename in .vscode is STM32L4x6.svd. On running update.py again, I get the following prompt:

```
Invalid path detected in 'buildData.json' 'stm32SvdPath' key.

        Enter path or command for 'stm32SvdPath':
        Paste here and press Enter: 

```
The path in my buildData.json is:

`"stm32SvdPath": ".vscode/STM32L4x6.svd",`

The original path is:

`/home/niek/Documents/code/KeilSVD/CMSIS/SVD/STM32L4x6.svd`

The rest of buildData.json:

```
    "gccExePath": "/usr/bin/arm-none-eabi-gcc",
    "gccInludePath": "/usr/lib/gcc/arm-none-eabi/8.2.1/include",
    "buildToolsPath": "/usr/bin/make",
    "pythonExec": "python3",
    "openOcdPath": "/usr/bin/openocd",
    "openOcdInterfacePath": "/usr/share/openocd/scripts/interface/stlink-v2-1.cfg"
```

schperplata, 2019-06-20:
Cannot reproduce. A couple of other things:
* By sharing an envrionment I meant: can you zip you project directory and share it? Are you sure you are using the latest ideScripts version?
* Can you share a log (terminal print statements from the beginning to the end)?
* Did you check if there are any access/permission issues within your directory? 
* Do the same thing happen if you run updatePaths.py before update.py?

I need more info to continue. This also might be OS issue, since everything works for me.

niekez, 2019-06-23:
Here you go, was to large for github: [download](https://we.tl/t-6G2GieZv5W)

My log:

```
 ~/Documents/code/base_stm32l496Gdisc_make  python ./ideScripts/update.py 
Update started.

Existing '.vscode' folder used.
One STM32CubeMX file found: base_stm32l496Gdisc_make.ioc
Existing 'Makefile' file found (restored from '.backup').
Copy of file (new name: Makefile): Makefile.backup
Makefile 'print-variable' function added.
Valid 'buildData.json' file found.
Valid 'toolsPaths.json' file found.
'toolsPaths.json' file updated!

Invalid path detected in 'buildData.json' 'stm32SvdPath' key.

        Enter path or command for 'stm32SvdPath':
        Paste here and press Enter: /home/niek/Documents/code/KeilSVD/CMSIS/SVD/STM32L4x6.svd
Existing 'c_cpp_properties.json' file found.
        Keys updated according to the template.
'c_cpp_properties.json' file updated!
New Makefile data succesfully written.
'buildData.json' file updated!
Build folder already exist: 'build'
Existing 'tasks.json' file found.
'tasks.json' file updated!
Existing 'launch.json' file found.
'launch.json' file updated!
Existing base_stm32l496Gdisc_make.code-workspace file found.
'*.code-workspace' file updated!

OK (36 seconds).

 ~/Documents/code/base_stm32l496Gdisc_make  python ./ideScripts/update.py
Update started.

Existing '.vscode' folder used.
One STM32CubeMX file found: base_stm32l496Gdisc_make.ioc
Existing 'Makefile' file found (restored from '.backup').
Copy of file (new name: Makefile): Makefile.backup
Makefile 'print-variable' function added.
Valid 'buildData.json' file found.
Valid 'toolsPaths.json' file found.
'toolsPaths.json' file updated!

Invalid path detected in 'buildData.json' 'stm32SvdPath' key.

        Enter path or command for 'stm32SvdPath':
        Paste here and press Enter: /home/niek/Documents/code/KeilSVD/CMSIS/SVD/STM32L4x6.svd
Existing 'c_cpp_properties.json' file found.
        Keys updated according to the template.
'c_cpp_properties.json' file updated!
New Makefile data succesfully written.
'buildData.json' file updated!
Build folder already exist: 'build'
Existing 'tasks.json' file found.
'tasks.json' file updated!
Existing 'launch.json' file found.
'launch.json' file updated!
Existing base_stm32l496Gdisc_make.code-workspace file found.
'*.code-workspace' file updated!

OK (5 seconds).
```
Permissions:
```
ls -la .
total 136
drwxrwxr-x 9 niek niek  4096 23 jun 11:00 .
drwxrwxr-x 8 niek niek  4096 23 jun 10:44 ..
-rw-r--r-- 1 niek niek   203 23 jun 11:02 base_stm32l496Gdisc_make.code-workspace
-rw-r--r-- 1 niek niek 29661 18 jun 10:47 base_stm32l496Gdisc_make.ioc
drwxr-xr-x 2 niek niek 12288 23 jun 11:01 build
drwxrwxr-x 4 niek niek  4096 18 jun 10:47 Drivers
drwxrwxr-x 3 niek niek  4096 18 jun 10:50 ideScripts
drwxrwxr-x 2 niek niek  4096 18 jun 10:47 Inc
-rw-r--r-- 1 niek niek  6653 23 jun 11:02 Makefile
-rw-r--r-- 1 niek niek  6978 18 jun 10:51 Makefile.backup
drwxrwxr-x 3 niek niek  4096 18 jun 10:47 Middlewares
-rw-r--r-- 1 niek niek 13867 18 jun 10:47 .mxproject
drwxrwxr-x 2 niek niek  4096 18 jun 10:47 Src
-rw-r--r-- 1 niek niek 14951  5 apr 16:45 startup_stm32l496xx.s
-rw-r--r-- 1 niek niek  6067 18 jun 10:47 STM32L496AGIx_FLASH.ld
drwxrwxr-x 2 niek niek  4096 18 jun 10:52 .vscode
```
Update Paths output:

```
python ./ideScripts/updatePaths.py                                                                                  1 
Existing '.vscode' folder used.
One STM32CubeMX file found: base_stm32l496Gdisc_make.ioc
Valid 'buildData.json' file found.
Valid 'toolsPaths.json' file found.

Valid path(s) for gccExePath detected: '/usr/bin/arm-none-eabi-gcc'.
        Update? [y/n]: n

Valid path(s) for buildToolsPath detected: '/usr/bin/make'.
        Update? [y/n]: n

Valid path(s) for pythonExec detected: 'python3'.
        Update? [y/n]: n

Valid path(s) for openOcdPath detected: '/usr/bin/openocd'.
        Update? [y/n]: n

Valid path(s) for openOcdInterfacePath detected: '/usr/share/openocd/scripts/interface/stlink-v2-1.cfg'.
        Update? [y/n]: n
'toolsPaths.json' file updated!
Traceback (most recent call last):
  File "./ideScripts/updatePaths.py", line 245, in <module>
    buildData = bData.prepareBuildData(request=True)
  File "/home/niek/Documents/code/base_stm32l496Gdisc_make/ideScripts/updateBuildData.py", line 119, in prepareBuildData
    buildData = paths.verifyTargetConfigurationPaths(buildData, request)
  File "/home/niek/Documents/code/base_stm32l496Gdisc_make/ideScripts/updatePaths.py", line 155, in verifyTargetConfigurationPaths
    msg = "\n\nValid path(s) for " + pathName + " detected: '" + buildData[pathName] + "'."
TypeError: can only concatenate str (not "list") to str
```
Same error when running update.py after.

Hope this helps.

schperplata, 2019-06-23:
Yeah, I've discovered the issue when running _updatePaths.py_ when I first try to reproduce your issue - this  indeed was a bug and was immediately fixed.
Still, shouldn't affect update.py procedure and the error you were describing before (unsaved new path to _.svd_ file - still unresolved).

Before we proceed, please update ideScripts folder to the latest revision and try again. I works perfectly fine on my PC with latest VS Code and python 3.6, 64bit.

niekez, 2019-06-23:
Updated, updatePaths error is gone. Svd path error remains.

schperplata, 2019-06-25:
So, since I can't reproduce issue, and I am not planing to install your OS just for this bug, you are on your own. But hey, you use VS Code and I imagine you are comfortable with a little coding if you use Archlinux :)

Here are breakpoints you can set and inspect what build data is read and hopefully you will find what/why this specific field gets corrupted. Select **Debug current Python file** launch configuration, set breakpoints and run (launhc debug session) on **update.py** file

Breakpoints:
_updateBuildData.py @ 98_ - check what buildData is read at the beginning. Try to directly modify some field and check (before debug), if you get the right build data dictionary.

_updateBuildData.py @ 122_ - check returned buildData on function exit. If at that point buildData is correct, move on, otherwise, step repeat the process and inspect where your buildData gets corrupted.

_updatePaths.py @ 151_ - this is the location of your error. Inspect what `buildData[pathName]` is in that moment

_updateBuildData.py @ 337_ - this is an actual write to _buildData.json_ file. Check what buildData is at that point and compare how it is written to the file. If everything seems fine, I would slowsly debug (step) through `prepareBuildData()` function  (in _updateBuildData.py_).

Let me know if you have further questions or if you need aditional help. I assume there will be something silly, or some OS specific issues (like paths or invalid characters). Hope you solve this issue.

niekez, 2019-06-30:
I may have found the problem, not a solution though. I noticed a difference between the openOcdconfig and svd path in buildData.json:

```
    "openOcdConfig": [
        ".vscode/stm32l4x.cfg"
    ],
    "stm32SvdPath": ".vscode/STM32L4x6.svd",
```
Changing it to:
```
    "openOcdConfig": [
        ".vscode/stm32l4x.cfg"
    ],
    "stm32SvdPath": [
        ".vscode/STM32L4x6.svd"
    ],

```
Prevents the error from occurring.

niekez, 2019-07-01:
It was checking a correct path to be a command, only in the case of a single string, not a list (that's why the ocd config didn't give the same error). I think the functionality was supposed to check for a command when the path was invalid, not a valid path.

What I don't get is why you didn't get the same error as me.

schperplata, 2019-07-07:
Yeah, this was a bug. I've fixed this and some other stuff with latest commit. Thank you.

Anyway, it is really annoying not knowing why I can't reproduce this on my own. Does Archlinux has any problems with paths/folders/files that start with `.`, like `.vscode/something.svd`?

I hope you can work further, and we will solve this issue once we have a more reliable way to reproduce it.

closed 2019-07-07 by niekez