damogranlabs/VS-Code-STM32-IDE #25: how to install

tdjastrzebski, 2019-06-22:
Is this product available as VC Code plugin? If so what is its name?
All I am finding in marketplace is "stm32-for-vscode" by Bureau Moeilijke Dingen

schperplata, 2019-06-22:
Ha, interesting, at the time I published this project, plugin you've recommended didn't exist yet.

Anyway, no. This is not a an extension. This is just a pack of scripts which prepare all the necessary files/settings for you to use VS Code as STM32 IDE. See [Setup procedure in README](https://github.com/damogranlabs/VS-Code-STM32-IDE/blob/master/README.md#setup).

tdjastrzebski, 2019-06-29:
Consider packaging this as VS Code Extension would make your solution way more easier to find.

schperplata, 2019-07-07:
Yeah, I am thinking about it for a while. Maybe later as a cold-winter-days project.
Another obstacle is that I still want _ideScripts_ to be customizable, which is harder to achieve with VS Code packed extension.

schperplata, 2020-04-20:
No further plans of upgrading this project at the moment.

closed 2020-04-20 by schperplata