damogranlabs/VS-Code-STM32-IDE #27: Building issue

Przasa, 2019-07-18:
Hi. I encountered a problem during building the code. It's seems, I am not able to lounch the "make.exe". Probably there is something missing in json.files, but I can't find the reason (everything look fine). 

I was able to run "make.exe" from cmd but, i don't know how to move forward with VS.
I'll be really grateful for receiving an advice. 

> Executing task in folder p0_starter: """C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.12-20190422-1053/bin/make.exe" "GCC_PATH=C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/armgcc/bin" -j6"" <

```
'""C:' is not recognized as an internal or external command,
operable program or batch file.
The terminal process command 'C:\Windows\system32\cmd.exe /d /c """C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.12-20190422-1053/bin/make.exe" "GCC_PATH=C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/armgcc/bin" -j6""' failed to launch (exit code: 1)

Terminal will be reused by tasks, press any key to close it.
```
```
{
    "env": {
        "____________________USER_FIELDS_CAN_BE_MODIFIED____________________": "",
        "user_cSources": [],
        "user_asmSources": [],
        "user_ldSources": [],
        "user_cIncludes": [],
        "user_asmIncludes": [],
        "user_ldIncludes": [],
        "user_cDefines": [],
        "user_asmDefines": [],
        "user_cFlags": [],
        "user_asmFlags": [],
        "user_ldFlags": [],
        "____________________DO_NOT_MODIFY_FIELDS_BELOW____________________": "",
        "cubemx_sourceFiles": [
            "Src/main.c",
            "Src/stm32f1xx_it.c",
            "Src/stm32f1xx_hal_msp.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio_ex.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_uart.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc_ex.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dma.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_cortex.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pwr.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c",
            "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash_ex.c",
            "Src/system_stm32f1xx.c",
            "startup_stm32f103xb.s"
        ],
        "cubemx_includes": [
            "Inc",
            "Drivers/STM32F1xx_HAL_Driver/Inc",
            "Drivers/STM32F1xx_HAL_Driver/Inc/Legacy",
            "Drivers/CMSIS/Device/ST/STM32F1xx/Include",
            "Drivers/CMSIS/Include",
            "Drivers/CMSIS/Include"
        ],
        "cubemx_defines": [
            "USE_HAL_DRIVER",
            "STM32F103xB",
            "USE_HAL_DRIVER",
            "STM32F103xB"
        ],
        "gccExePath": "C:\\Users\\Piotr\\AppData\\Roaming\\GNU MCU Eclipse\\ARM Embedded GCC\\armgcc\\bin\\arm-none-eabi-gcc.EXE",
        "gccIncludePath": "C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/armgcc/lib/gcc/arm-none-eabi/8.3.1/include"
    },
    "configurations": [
        {
            "name": "p0_starter_workspace",
[... 149 lines cut ...]
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "cwd": "${workspaceFolder}",
            "executable": "build/p0_starter.elf",
            "svdFile": ".vscode/STM32F103xx.svd",
            "configFiles": [
                "C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/OpenOCD/0.10.0-12-20190422-2015/scripts/target/stm32f1x_stlink.cfg",
                ".vscode/stm32f1x.cfg"
            ],
            "preLaunchTask": "Build project"
        },
        {
            "name": "Debug current Python file",
            "type": "python",
            "request": "launch",
            "cwd": "${workspaceFolder}",
            "program": "${file}",
            "console": "integratedTerminal"
        }
    ]
}
```
[image]
[image]

schperplata, 2019-07-18:
Hello Przasa!
Before we continue with resolving this issue, just try to change VS Code default terminal to PowerShell.

If I use `cmd`, I get the same error:
```
''C:' is not recognized as an internal or external command,
operable program or batch file.
```

Let me know.

Przasa, 2019-07-19:
Hello shperplata
Changing the the terminal to powershell helped to build the the code --> thanks for that.
However i still have problems with including files. I can imagine including files manually, but it doesn't seems to be correct way. 
(Folders' structure looks fine in my opinion)

[image]
[image]

Moreover, building the project the second (and nth) time, leads to some problems:
```
> Executing task in folder p0_starter: & 'C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/Build Tools/2.12-20190422-1053/bin/make.exe' 'GCC_PATH=C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/ARM Embedded GCC/armgcc/bin' -j6 <

make: Nothing to be done for 'all'.

Terminal will be reused by tasks, press any key to close it.

> Executing task in folder p0_starter: & 'C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/OpenOCD/0.10.0-12-20190422-2015/bin/openocd.exe' -f 'C:/Users/Piotr/AppData/Roaming/GNU MCU Eclipse/OpenOCD/0.10.0-12-20190422-2015/scripts/target/stm32f1x_stlink.cfg' -f .vscode/stm32f1x.cfg -c 'program build/p0_starter.elf verify reset exit' <

GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: target/stm32f1x_stlink.cfg is deprecated, please switch to target/stm32f1x.cfg
Error: Debug adapter does not support any transports? Check config file order.
Error: unable to select a session transport. Can't continue.
shutdown command invoked

Terminal will be reused by tasks, press any key to close it.
```

I would be grateful for receiving advise.

Best Regards,
PRzasa

schperplata, 2019-07-19:
>Changing the the terminal to powershell helped to build the the code --> thanks for that.

Great, I will add this to the docs.

>However i still have problems with including files. I can imagine including files manually, but it doesn't seems to be correct way. (Folders' structure looks fine in my opinion)

I can't really see what is your complete folder structure. What is your workspace (current) working directory? 'c_cpp_properties.json' paths should be relative to cwd - workspace root directory.

> Moreover, building the project the second (and nth) time, leads to some problems:
make: Nothing to be done for 'all'.

This is not an error, just a notification of make.exe - nothing to do, project is already in the latest build state.

>WARNING: target/stm32f1x_stlink.cfg is deprecated, please switch to target/stm32f1x.cfg

As it says, change to correct interface, and see if other errors will be resolved. Following lines (`Error: Debug adapter does not support any transports? Check config file order. Error: unable to select a session transport. Can't continue.`) might also be connection/board issues.

Also, switch to previous version of compiler. I didn't check the latest version, but there were some issues reported [here](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/9#issuecomment-463194218).

[image]

Przasa, 2019-07-20:
I have followed your instruction, but i still struggling with files linking. 

After implementing basic instruction to main.c, VSCode is showing some errors also.
[image]

Also the cortex dubugger is not able to run:
[image]

I am sending the full project here. Please help me to find the issue.
https://gofile.io/?c=vAbRQA

schperplata, 2019-07-20:
>After implementing basic instruction to main.c, VSCode is showing some errors also.

Yeah, I see the problem now. I wasn't paying attention before, since my autocomplete/go to definition worked. What you are seeing is [already reported VS Code bug (#3912)](https://github.com/microsoft/vscode-cpptools/issues/3912), but it shouldn't affect the build process. IDE indeed shows error squigles, but paths are actually resolved when build process is run. Is it true, your build task is successfully executed?

> Also the cortex dubugger is not able to run

This issue is a little more complex as it seems, and I can't give you instructions how to solve it - I don't have any STM32F**1** board. Anyway, don't give up yet, here are a couple of clues:
* As it seems to me, this is an openod device description file issue (_stm32f1x.cfg_ in your case). I tried to change all CPU definitions (defines and .cfg files) with STM32F0 (a board that I have) and everything worked as expected.
* I see that STM32F103xB is selected in _Makefile.backup_ generated by CubeMX. Make sure _stm32f1x.cfg_  is the right file for this CPU - I googled and found this [this](https://buger.dread.cz/category/stm32.html) and [this](https://github.com/texane/stlink/issues/545) issues where this CPUs can't be flashed.
* A little more descriptive error can be achieved by running some of **CPU: \*** tasks. If I run **CPU: Run** task, this is displayed:
[image]
Line: _Warn : UNEXPECTED idcode: 0x0bb11477_ makes sense, since I don't have STM32F0 device connected. My point is, run this task and inspect the output. 
According to .cfg files, STM32F0 has only SWD, while STM32F1x has SWD and JTAG. Probably OpenOCD+stlink are not configured properly and some additional setup is required, as for example in _stm32f0discovery.cfg_. 

Just as a precaution: as *.json fields can be edited manually, it is easy to make mistake and not update all the relevant fields in all *.json files. The safest way is to just use updatePaths.py script while changing workspace settings.

Przasa, 2019-07-29:
Thank you for spending time for my case.

I'll try to expirements with your clues.  In the meanwhile i switched to platform.io where everything works fine. Since it's a pretty similar project, i should compare the differences between them.

If i'll get the solution, I'll share it hare.

closed 2019-07-29 by Przasa