damogranlabs/VS-Code-STM32-IDE #28: .vscode cfg file not found by openocd

niekez, 2019-07-26:
When I start debugging it times out. When I run the command manually I get this output:

```
$ "/home/niek/opt/npm/bin/openocd" "-c" "gdb_port 50000" "-s" "/home/niek/Documents/code/base/disc_spi" "-f" "/home/niek/opt/xPacks/@gnu-mcu-eclipse/openocd/0.10.0-12.1/.content/scripts/interface/stlink.cfg" "-f" ".vscode/stm32l4x.cfg"
```

```
"/home/niek/opt/npm/bin/openocd" "-c" "gdb_port 50000" "-s" "/home/niek/Documents/code/base/disc_spi" "-f" "/home/niek/opt/xPacks/@gnu-mcu-eclipse/openocd/0.10.0-12.1/.content/scripts/interface/stlink.cfg" "-f" ".vscode/stm32l4x.cfg"
```

When I change the launch.json to the following:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex debug",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "cwd": "${workspaceFolder}",
            "executable": "build/disc_spi.elf",
            "svdFile": ".vscode/STM32L4x6.svd",
            "showDevDebugOutput": true,
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32l4x.cfg"
            ],
            "preLaunchTask": "Build project"
        },
        {
            "name": "Debug current Python file",
            "type": "python",
            "request": "launch",
            "cwd": "${workspaceFolder}",
            "program": "${file}",
            "console": "integratedTerminal"
        }
    ]
}
```
It's working fine.

The problem I think lies in the fact it can't find the local cfg file. I would like to be able to the relative paths like the above but these aren't accept by updatePaths.py.

schperplata, 2019-07-26:
Hmm. It would be very helpful if there would be other non-windows users - I am having a hard time since most of this issues are somehow linux related and I can't reproduce them. So, linux guys, speak up! :)

Anyway, paths can be relative. But, if file on a given relative path does not exist, this path is not a valid path and is rejected by the script.
I don't know for sure if this is the problem, but still, what is this **@** in your path?
```"... "-f" "/home/niek/opt/xPacks/@gnu-mcu-eclipse/openocd/0.10.0-12.1/.content/scripts/interface/stlink.cfg" ...```?
Also, I use OpenOCD version 0.10.0-8-20180512-1921, didn't check with your newer one.

niekez, 2019-07-27:
For me the debugging at first didn't work with the openocd version from the system repository. 

The relative paths are not accepted because the binary for openocd is in /usr/bin/openocd and the scripts folder is in /usr/share/openocd/. I can get around this by entering absolute paths.

> I don't know for sure if this is the problem, but still, what is this @ in your path?
I used xpm to install the gnu-mcu-eclipse version of openocd, not ideal because it has version numbering in in the path.

I just found out that it didn't work because the cfg file in my .vscode folder was the system one, not gnu-mcu-eclipse one. So the problem wasn't not finding it but having a mismatch in version.

Using the gnu-eclipse version like recommended by cortex-debug works for me after replacing the target cfg in .vscode.

niekez, 2019-07-30:
So in the end I used the latest release from here: https://github.com/gnu-mcu-eclipse/openocd/releases

And had to remove the target cfg from .vscode en update with the cfg files from gnu-mcu-eclipse. This also has an updated interface stlink.cfg which archlinux' openocd from the repository doesn't have.

closed 2019-07-30 by niekez