damogranlabs/VS-Code-STM32-IDE #31: GNU eclipse tools, GNU Eclipse ARM Embedded GCC: 7-2018-q2 not available

AO111497, 2019-10-10:
The link provided to download the GNU Eclipse ARM Embedded GCC version 7-2018-q2 does not work. Is v7.2.1 the desired version or is there another version that will work?

schperplata, 2019-10-10:
As it seems, link and releases site was deprecated, moved to GitHub.
I've updated the docs and successfully tested the example projects with the latest tools (GCC, make, OpenOCD). 

For those who are switching to newer versions of tools: to update current paths in you project (and global VS Code STM32 IDE toolsPaths.json), use 'updatePaths.py' script.

Thank you, let me know if it works for you.

AO111497, 2019-10-10:
Thank you, I downloaded v7.2.1. Now I am following the instruction given in the video on how to get started, but I keep getting the following message:
 Python version 3 or later installation not detected, please install or enter custom path/command below.

I have python 3.3.3 installed and provide the directory: C:\Python33\python.exe
But it gives me the following error:

ERROR (766 seconds).
Unexpected error occured during 'Update' procedure. Exception:
Traceback (most recent call last):
  File "c:/Users/Documents/VS/VS-Code-STM32-IDE-master/example/STM32F051K4/ideScripts/update.py", line 82, in <module>      
    tasksData = tasks.addAllTasks(tasksData)
  File "c:\Users\Documents\VS\VS-Code-STM32-IDE-master\example\STM32F051K4\ideScripts\updateTasks.py", line 127, in addAllTasks
    task = self.getBuildTask()
  File "c:\Users\Documents\VS\VS-Code-STM32-IDE-master\example\STM32F051K4\ideScripts\updateTasks.py", line 212, in getBuildTask
    numOfCores = os.cpu_count()
AttributeError: 'module' object has no attribute 'cpu_count'

Am I referencing the wrong python? Am I supposed to give the path to the add on in vs code? I have no idea where that file would be located.

schperplata, 2019-10-11:
Use Python 3.6 or newer. 
Which Python VS Code will use [should be selected as usual when using VS Code](https://code.visualstudio.com/docs/python/environments).

closed 2020-04-20 by schperplata