damogranlabs/VS-Code-STM32-IDE #32: Problem with FreeRTOS

NN708, 2019-10-26:
I'm trying to compile a project with FreeRTOS enabled, but the compilation failed when `cmsis_os.h` and `FreeRTOS.h` can't be found. These files are generated by STM32CubeMX and are stored in `Middlewares/Third_Party/FreeRTOS/Source`. It seems that VS-Code-STM32-IDE didn't look into the folder.

Here's the hierarchy:
[image]

schperplata, 2019-10-26:
Did you check makefile - is folder path `Middlewares/Third_Party/FreeRTOS/Source` listed in 
```
# C includes
C_INCLUDES =
```
and source files added in
```
# C sources
C_SOURCES =
```
...? If not, try to delete Makefile, and re-generage project again with CubeMX. 
If these paths are added, check if they are listed in `c_cpp_properties.json` and `buildData.json`, and report back. Consider sharing an example code/project.

NN708, 2019-10-27:
I deleted and then regenerated Makefile and it worked. Thank you!

closed 2019-10-27 by NN708