damogranlabs/VS-Code-STM32-IDE #33: Maybe I got a bug.

UPUPGOO, 2019-10-31:
First of all, thank you for your tool. It is really helpful.
I want to debug STM32F407ZET6, but I got one error.
[image]
The console outputted 

> Please check OUTPUT tab (Adapter Output) for output from openocd
Launching server: "openocd" "-c" "gdb_port 50000" "-s" "C:\Users\ShengJiaPeng\Documents\PlatformIO\Projects\STM32F4" "-f" "interface/stlink-v2-1.cfg" "-f" "target/stm32f4x.cfg.cfg"

This is launch.json:

> 
      "showDevDebugOutput": true,
      "cwd": "${workspaceRoot}",
      "executable": "./build/STM32F4.elf",
      "name": "Debug STM32",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "preLaunchTask": "Build STM",
      "device": "stlink",
      "configFiles": [
        "interface/stlink-v2-1.cfg",
        "target/stm32f4x.cfg.cfg"
      ]

When I changed  "target/stm32f4x.cfg.cfg" into "target/stm32f4x_stlink.cfg", everything was good.

schperplata, 2019-10-31:
```"configFiles"``` section and its values is created on _update.py_ run. On first run you are asked to specify target configuration files (and other tool/file paths), so I assume you made a mistake at the beginning. This paths are than cached in _buildData.json._ and you don't need to set them again, unless qou manually run _updatePaths.py._

Also, check that you are using the latest version. Latest version create a copy of cfg files in _.vscode_ folder.

UPUPGOO, 2019-11-01:
> On first run you are asked to specify target configuration files (and other tool/file paths)

Sorry, I was not asked to specify any configuration file. The program was completely automatical to generate configuration files. I just ran 'Build STM32 project' command.

schperplata, 2019-11-01:
Are you sure you have reported a bug at the correct place?
A couple of things that are strange and/or not the same as in my default scripts: 
* There is no 'Build STM project' task. There is only 'Build project' task.
* In tasks there is no: ```"preLaunchTask": "Build STM",```
* ```PlatformIO\Projects\STM32F4.``` Are you using ```PlatformIO``` or ```stm32-for-vscode``` extension?

UPUPGOO, 2019-11-01:
sorry, I made a mistake. I was using stm32-for-vscode extension.  I mixed up these two giuhub projects between stm32-for-vscode  and yours because of the similar name. I am sorry again.

closed 2019-11-01 by UPUPGOO