damogranlabs/VS-Code-STM32-IDE #34: Problem with access (seemingly) when executing update.py

fstkmaro, 2019-11-07:
First, thanks for coming up with this amazing solution. I have installed everything as required(although I had to modify some of the structures of the tool folders etc). I can run update.py and provide the paths to the requested directories. I have placed these directories in the recommended location "c:/users/user/app_data/roaming/GNU*Eclipse" In this directory, I have all other folders in the instructions. One thing I noticed is that when I open the folder in vsc, it does not do that as a workspace, I have to save it as one in order for the *.workspace file to be created which is required by the script. Anyways, I get to the point where I specify the STM32*.cfg file and after that, I need to specify the svd file to use. I do that and I get errors seemingly having to do with permissions. Before I did anything to change the permissions on my machine, I wanted to be sure that it has nothing to do with the location chosen for the files. The output of update.py is as follows: 

> 
"Update started.

Existing '.vscode' folder used.
One STM32CubeMX file found: L476VSCTest.ioc
Existing 'Makefile' file found (restored from '.backup').
Copy of file (new name: Makefile): d:/Development/STM32/cubemxprojects/L476VSCTest/Makefile.backup
Makefile 'print-variable' function added.
Valid 'buildData.json' file found.
Valid 'toolsPaths.json' file found.
'toolsPaths.json' file updated!

        Enter path(s) to OpenOCD configuration file(s):
                Example: 'target/stm32f0x.cfg'. Absolute or relative to OpenOCD /scripts/ folder.
Traceback (most recent call last):
  File "d:/Development/STM32/cubemxprojects/L476VSCTest/ideScripts/update.py", line 56, 
in <module>
    makefileData = makefile.getMakefileData(makeExePath, gccExePath)
  File "d:\Development\STM32\cubemxprojects\L476VSCTest\ideScripts\updateMakefile.py", line 93, in getMakefileData
    projectName = self.getMakefileVariable(makeExePath, gccExePath, self.mkfStr.projectName)[0]
  File "d:\Development\STM32\cubemxprojects\L476VSCTest\ideScripts\updateMakefile.py", line 366, in getMakefileVariable
    proc = Popen(arguments, stdout=PIPE)
  File "C:\Users\omis2\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\omis2\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
PermissionError: [WinError 5] Access is denied"

Any help is appreciated. By the way, the machine is a Windows 10 Pro machine.

fstkmaro, 2019-11-07:
I have been playing around. Running other unrelated python scripts may yield the same 'access denied" result unless python is being run as an admin. Running vsc as admin results in the same. I just thought I'd share. thanks

schperplata, 2019-11-07:
> One thing I noticed is that when I open the folder in vsc, it does not do that as a workspace, I have to save it as one in order for the .workspace file to be created which is required by the script.
This is the default behaviour of VS Code. According to the docs, user must first save opened folder as workspace to continue.

Anyway, thanks for sharing the observation, so others can also benefit from your findings! 
Check your general windows user account privileges for system (and other) folders. It must be some system-wide misconfiguration if other python scripts are also not working.

Good luck!

kumarmohan7, 2019-11-24:
this Error how to solve 
PermissionError: [WinError 5] Access is denied

kumarmohan7, 2019-11-24:
Traceback (most recent call last):
  File "c:\Users\kumar\.vscode\extensions\ms-python.python-2019.11.50794\pythonFiles\ptvsd_launcher.py", line 43, in <module>
    main(ptvsdArgs)
  File "c:\Users\kumar\.vscode\extensions\ms-python.python-2019.11.50794\pythonFiles\lib\python\old_ptvsd\ptvsd\__main__.py", line 432, in main      
    run()
  File "c:\Users\kumar\.vscode\extensions\ms-python.python-2019.11.50794\pythonFiles\lib\python\old_ptvsd\ptvsd\__main__.py", line 316, in run_file
    runpy.run_path(target, run_name='__main__')
  File "C:\Users\kumar\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 262, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\kumar\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 95, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\kumar\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 501, in <module>
    makefileData = makefile.getMakefileData(makeExePath, gccExePath)
  File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 93, in getMakefileData
    projectName = self.getMakefileVariable(makeExePath, gccExePath, self.mkfStr.projectName)[0]
  File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 366, in getMakefileVariable
    proc = Popen(arguments, stdout=PIPE)
  File "C:\Users\kumar\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\kumar\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
  File "c:\Users\kumar\.vscode\extensions\ms-python.python-2019.11.50794\pythonFiles\lib\python\old_ptvsd\ptvsd\_vendored\pydevd\_pydev_bundle\pydev_monkey.py", line 480, in new_CreateProcess
    return getattr(_subprocess, original_name)(app_name, patch_arg_str_win(cmd_line), *args)
PermissionError: [WinError 5] Access is denied

schperplata, 2019-11-24:
@kumarmohan7  did you read the topic from the beginning? Did you try running other scripts? did you try anything beside what you [already reported](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/35), and now duplicated here?
Point is, just by copying error is not enought for anyone to help you. Make everybody a favor and precisely describe and try:
* running other non-ideScripts py scripts from the same workspace
* check what is your terminal using (cmd/powershell)
* does directory actually have R/W permissions?
* do you have the latest ideScripts?
* try to debug this script, that is why you have "Debug current Python file" configuration:
set breakpoint at and explore further:
_File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 366, in getMakefileVariable_
***

fstkmaro, 2019-12-03:
> @kumarmohan7  did you read the topic from the beginning? Did you try running other scripts? did you try anything beside what you already reported, and now duplicated here?
> Point is, just by copying [...]

Thank you for getting back on this topic. If you read my last reply to the issue, the behavior happens regardless of user or superuser being used on a Windows machine. A script written in python and run out of vscode does not seem to cause trouble. What seems to be the difficulty is that Windows may be reacting erroneously when a directory is being modified. Me being new to Python, am having difficulty understanding the issue. All other parts of the execution of the provided script seem to work well except for that last part after entering the required toolchain targets. For clarity, I am running Windows 10 Pro for workstations version 1903. I have yet to try running this on a different PC. I have setup myself as a system wide admin to no avail. Thanks again for your time.

schperplata, 2019-12-03:
What is your python version? I am currently running python 3.7.
1. Do you have only one python installed on your PC - does VS Code use exactly the same  python.exe as you do when you run your script outside of VS Code?
2. Did you try to run the script without debug session (eg, right click in explorer pane, "Run Python file in Terminal")?
3.Try to close all program that could possibly lock this file (CubeMX, external editor that has opened Makefile, ...). 
4. Try to debug it with VS Code: just set breakpoint: 
```
File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 93, in getMakefileData
projectName = self.getMakefileVariable(makeExePath, gccExePath, self.mkfStr.projectName)[0]
```
... and step line by line to see what are the arguments and if time (race conditions?) occure?

Also, create a manual backup of your project, delete all makefiles, reopen CubeMX and re-generate project. Make sure you have a valid Makefile.
I am also having problems with CubeMX Makefile handling magic, still investigating - but your problem might be unrelated.

fstkmaro, 2019-12-04:
> What is your python version? I am currently running python 3.7.
> 
> Do you have only one python installed on your PC - does VS Code use exactly the same  python.exe as you do when you run your script outside of VS Code?
> Did you try to run the script without debug session (eg, right click in explorer pane, "Run Python file in Terminal")?
> 3.Try to close all program that could possibly lock this file (CubeMX, external editor that has opened Makefile, ...).
> Try to debug it with VS Code: just set breakpoint:
> 
> File "d:\STM32\VSCode_STM32_IDE\STM32CubeMx_WS\STM32_Test\ideScripts\updateMakefile.py", line 93, in getMakefileData
> projectName = self.getMakefileVariable(makeExePath, gccExePath, self.mkfStr.projectName)[0]
> 
> ... and step line by line to see what are the arguments and if time (race conditions?) occure?
> Also, create a manual backup of your project, delete all makefiles, reopen CubeMX and re-generate project. Make sure you have a valid Makefile.
> I am also having problems with CubeMX Makefile handling magic, still investigating - but your problem might be unrelated.

Thank you. The answers to your questions, I am using 3.8 now but have tried it with 3.7 and yes, it happens even running in terminal. I will look into #3 as I'd like a fresher output. I will keep you posted. 
Thanks again.

closed 2020-04-20 by schperplata