damogranlabs/VS-Code-STM32-IDE #38: gccExePath

m1ke02, 2020-05-10:
Hi, 

What should be the value of "gccExePath"?
Must it be "C:/Programs/GNU_arm_tools/ARM_Embedded_GCC/bin/arm-none-eabi-gcc.exe" or 
"C:/Programs/GNU_arm_tools/ARM_Embedded_GCC/bin"?

If i choose last one, updatePaths.py fails to locate <stdint.h>.
If first, the script runs good, but VSCode displays a warning in c_cpp_properties.json "Path is not a directory: "C:\Programs\GNU_arm_tools\ARM_Embedded_GCC\bin\arm-none-eabi-gcc.exe""

BTW, there is a second warning "IntelliSense mode msvc-x64 is incompatible with compiler path."

schperplata, 2020-05-10:
> What should be the value of "gccExePath"?
Must it be "C:/Programs/GNU_arm_tools/ARM_Embedded_GCC/bin/arm-none-eabi-gcc.exe" or "C:/Programs/GNU_arm_tools/ARM_Embedded_GCC/bin"?

As you can see in [example code here](https://github.com/damogranlabs/VS-Code-STM32-IDE/blob/master/example/STM32F051K4/.vscode/buildData.json), this is a path to .exe file.

> If first, the script runs good, but VSCode displays a warning in c_cpp_properties.json "Path is not a directory: "C:\Programs\GNU_arm_tools\ARM_Embedded_GCC\bin\arm-none-eabi-gcc.exe""

Make sure you are updating the paths with _update.py_ script - do not manually edit paths in buildData/c_cpp_properties files. If you still won't be able to make it work, re-check your paths and make sure you have default folder structure as advised in readme.

> BTW, there is a second warning "IntelliSense mode msvc-x64 is incompatible with compiler path."

Wierd, I've never seen this warning. Currently I can't check since I just got a fresh PC, and I don't have all the tools yet. Do you have any special VS Code settings (intellisense, extensions, ...)?

m1ke02, 2020-05-10:
> Make sure you are updating the paths with _update.py_ script - do not manually edit paths in buildData/c_cpp_properties files. If you still won't be able to make it work, re-check your paths and make sure you have default folder structure as advised in readme.

Yes, I run update.py after updatePaths.py. And I made folder structure the same as default. It does not affect these warnings.

> 
> > BTW, there is a second warning "IntelliSense mode msvc-x64 is incompatible with compiler path."
> 
> Wierd, I've never seen this warning. Currently I can't check since I just got a fresh PC, and I don't have all the tools yet. Do you have any special VS Code settings (intellisense, extensions, ...)?

Nothing unusual, I have C/C++ 0.27.1, Python 2020.4.76186 and Cortex-Debug 0.3.6.
All I've done is switch default shell to PowerShell cause the scripts use single quotes in file names but Cmd doesn't understand them.

Version: 1.45.0 (user setup)
Commit: d69a79b73808559a91206d73d7717ff5f798f23c
Date: 2020-05-07T16:18:48.860Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18362