damogranlabs/VS-Code-STM32-IDE #39: Paths with spaces

m1ke02, 2020-05-21:
Hi!

In the readme you state that paths to compiler and build tools can contain spaces.
But for me build command generated by the script does not work (ran it both through VSCode task and manualy under Build Tools shell):

$ make "GCC_PATH=C:/Programs/GNU MCU Eclipse/ARM Embedded GCC/bin" -j6
make: *** [Makefile:156: build/main.o] Error 127
make: *** Waiting for unfinished jobs....
make: *** [Makefile:156: build/stm32f4xx_it.o] Error 127
c:/programs/gnu mcu eclipse/build tools/2.12-20190422-1053/bin/sh: C:/Programs/GNU: not found

Under Build Tools shell It can be fixed by adding extra quotes to GCC_PATH:
$ make "GCC_PATH='C:/Programs/GNU MCU Eclipse/ARM Embedded GCC/bin'" -j6
... Build succesful

But build task ran from VSCode does not work even after adding quotes to GCC_PATH in tasks.json:

> Executing task in folder Stm32CubeGCC: & 'C:/Programs/GNU MCU Eclipse/Build Tools/2.12-20190422-1053/bin/make.exe' GCC_PATH='C:/Programs/GNU MCU Eclipse/ARM Embedded GCC/bin' -j6 <

mkdir build
C:/Programs/GNU MCU Eclipse/ARM Embedded GCC/bin/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F429xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Src/main.c -o build/main.o
C:/Programs/GNU MCU Eclipse/ARM Embedded GCC/bin/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F429xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/stm32f4c:/programs/gnu mcu eclipse/build tools/2.12-20190422-1053/bin/sh: xx_it.d" -Wa,-C:/Programs/GNU: not found
a,-ad,-alms=build/stm32f4xx_it.lst Src/stm32f4xx_it.c -o build/stm32f4xx_it.o
make: *** [Makefile:156: build/main.o] Error 127
make: *** Waiting for unfinished jobs....
c:/programs/gnu mcu eclipse/build tools/2.12-20190422-1053/bin/sh: C:/Programs/GNU: not found
make: *** [Makefile:156: build/stm32f4xx_it.o] Error 127
The terminal process terminated with exit code: 1

Default shell set to PowerShell

schperplata, 2020-05-21:
> Default shell set to PowerShell

Did you try to use [cmd.exe instead of power shell](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/1#issuecomment-437684011)?

m1ke02, 2020-05-21:
> Did you try to use [cmd.exe instead of power shell](https://github.com/damogranlabs/VS-Code-STM32-IDE/issues/1#issuecomment-437684011)?

Yes, tried with no luck.
cmd.exe does not interpret single quote as a quote at all, just as a regular character.