damogranlabs/VS-Code-STM32-IDE #43: Try build and get this:  "undefined reference to ....

SergNif, 2020-11-02:
I am trying compile project. But get mistake.

I tried make simple project for stm32f103c8. It like for me. but this project for Oled 0.96 ssd1306 can`t to compile.

my project https://yadi.sk/d/TUBpLaJGUWKadQ
[Uploading mix.zip…]()
[image]

SergNif, 2020-11-02:
[image]

schperplata, 2020-11-03:
What you are seeing here is not related to ideScripts project, but a basic compile/link -time error. 
What linker (`ld.exe`) is telling you is: `undefined reference to 'SSD1306_Init'` which basically means that you are calling function `SSD1306_Init()` that is not defined or definition can't be discovered by the linker. Make sure you add all source files on correct places and includes are properly resolved.

Closing this issue as it is not related to ideScripts.

closed 2020-11-03 by schperplata