damogranlabs/VS-Code-STM32-IDE #44: update.py generates bad arguments for openocd

ng91, 2021-03-27:
update.py generates a directory instead of exactly path with file included - openocd says it's a directory not a file
[image]

schperplata, 2021-03-27:
What path did you specify when you run `update.py`? Can you share your `buildData.json`?

ng91, 2021-03-27:
`{
    "ABOUT1": "This file holds combined user and CubeMX generated Makefile workspace dependecies.",
    "ABOUT2": "User should not edit this fields, instead it should edit 'c_cpp_properties.json'",
    "ABOUT3": "This file is regenerated on 'Update workspace' task.",
    "VERSION": "1.7",
    "LAST_RUN": "2021-03-27 21:40:38.102041",
    "cSources": [
        "Src/main.c",
        "Src/usb_device.c",
        "Src/usbd_conf.c",
        "Src/usbd_desc.c",
        "Src/usbd_cdc_if.c",
        "Src/stm32f1xx_it.c",
        "Src/stm32f1xx_hal_msp.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio_ex.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd_ex.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_ll_usb.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc_ex.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dma.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_cortex.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pwr.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash_ex.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_exti.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_spi.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim.c",
        "Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c",
        "Src/system_stm32f1xx.c",
        "Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c",
        "Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c",
        "Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ioreq.c",
        "Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c"
    ],
    "asmSources": [],
    "ldSources": [
        "-lc",
        "-lm",
        "-lnosys"
    ],
    "cIncludes": [
        "Inc",
        "Drivers/STM32F1xx_HAL_Driver/Inc",
        "Drivers/STM32F1xx_HAL_Driver/Inc/Legacy",
        "Middlewares/ST/STM32_USB_Device_Library/Core/Inc",
        "Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc",
        "Drivers/CMSIS/Device/ST/STM32F1xx/Include",
        "Drivers/CMSIS/Include",
        "Drivers/CMSIS/Include"
    ],
    "asmIncludes": [],
    "ldIncludes": [],
    "cDefines": [
        "USE_HAL_DRIVER",
        "STM32F103xB"
    ],
    "asmDefines": [],
    "cFlags": [
        "-mcpu=cortex-m3",
        "-mthumb",
        "-DUSE_HAL_DRIVER",
        "-DSTM32F103xB",
        "-IInc",
        "-IDrivers/STM32F1xx_HAL_Driver/Inc",
        "-IDrivers/STM32F1xx_HAL_Driver/Inc/Legacy",
        "-IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc",
        "-IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc",
        "-IDrivers/CMSIS/Device/ST/STM32F1xx/Include",
        "-IDrivers/CMSIS/Include",
        "-IDrivers/CMSIS/Include",
        "-Og",
        "-Wall",
        "-fdata-sections",
        "-ffunction-sections",
        "-g",
        "-gdwarf-2",
        "-MMD",
        "-MP",
        "-MF"
    ],
    "asmFlags": [
        "-mcpu=cortex-m3",
        "-mthumb",
        "-Og",
        "-Wall",
        "-fdata-sections",
        "-ffunction-sections"
    ],
    "ldFlags": [
        "-mcpu=cortex-m3",
        "-mthumb",
        "-specs=nano.specs",
        "-TSTM32F103CBTx_FLASH.ld",
        "-lc",
        "-lm",
        "-lnosys",
        "-Wl,-Map=build/V1.map,--cref",
        "-Wl,--gc-sections"
    ],
    "buildDir": "build",
    "targetExecutablePath": "build/V1.elf",
    "cubeMxProjectPath": "V1.ioc",
    "openOcdConfig": [
        ".vscode/stm32f1x.cfg"
    ],
    "stm32SvdPath": ".vscode/STM32F103xx.svd",
    "ABOUT4": "---- Paths below are fetched from user-specific 'toolsPaths.json'. ----",
    "gccExePath": "/usr/bin/arm-none-eabi-gcc",
    "gccInludePath": "/usr/lib/gcc/arm-none-eabi/6.3.1/include",
    "buildToolsPath": "/usr/bin/make",
    "pythonExec": "python3",
    "openOcdPath": "/usr/bin/openocd",
    "openOcdInterfacePath": "/usr/share/openocd/scripts/interface"
}`
that is my buildData.json file. I see there is `"openOcdInterfacePath": "/usr/share/openocd/scripts/interface"`.
But when I change it to `/usr/share/openocd/scripts/interface/stlink-v2.cfg`, and run update.py, It is making this path   `"openOcdInterfacePath": "/usr/share/openocd/scripts/interface"` again, so nothing changes. Update.py doesn't asks me anything. When i ran it first time, It asked me for directories to stlink-v2.cfg, and i provided excatly  `"openOcdInterfacePath": "/usr/share/openocd/scripts/interface"`

ng91, 2021-03-27:
I used `updatePaths.py` to correct paths, and now when I run update.py, all directories are as i set, and flashing is fine! It's weird that I wrote the same path with file name included, but it didn't took into account the filename - `stlink-v2.cfg`.
Fortunately invoking `updatePaths.py` did the trick. Thanks!

closed 2021-03-27 by ng91